Seeed-Studio/LoRaWan-E5-Node #1: Clean project without replacing SDK files

jpmeijers, 2021-12-07:
In the readme it states:

> Download and copy this repo to your SDK folder en.stm32cubewl\STM32Cube_FW_WL_V1.0.0\Projects\NUCLEO-WL55JC\Applications\LoRaWAN and replace the origin en.stm32cubewl\STM32Cube_FW_WL_V1.0.0\Projects\NUCLEO-WL55JC\Applications\LoRaWAN\LoRaWAN_End_Node folder

This is not great.

Please change this repo to be a clean STM32CubeIDE project which one can clone to your workspace and open from there.

When one would develop firmware for the Lora-E5 you would not go and edit the STM32CubeSDK. You would create a new project which resided in the workspace. This example should be a starting point for new firmware, and therefore should be a clean project in the workspace.

jpmeijers, 2021-12-07:
Please have a look at https://github.com/androbi-com/LoRa-E5-Mini-EndNode, as that is what this official Seeed example should be like.

danak6jq, 2021-12-08:
I created:  https://github.com/danak6jq/Seeed-LoRa-E5  which is freshly generated to use the WLE5x and has BSP support for the LoRa-E5 Mini. It cleanly regenerates and builds. It may also be interesting.

Pillar1989, 2021-12-09:
@danak6jq I don't understand. You said your project clean regenerates and builds, can you help point out which files are regenerated and built?

danak6jq, 2021-12-09:
> @danak6jq I don't understand. You said your project clean regenerates and builds, can you help point out which files are regenerated and built?

I can run CubeMX, change configuration, and select "Generate code". Are you familiar with CubeMX (which is also part of STM32CubeIDE) ? Starting with V1.1.0 of ST's WLxx firmware package, Cube/CubeMX can be used to (re)generate LoRaWAN applications.

I started with no project at all, used "New > STM32 Project " and used STM32CubeIDE to create the LoRaWAN without copying or patching the LoRaWAN_End_Node example.

danak6jq, 2021-12-09:
@Pillar1989 the V1.0.0 LoRaWAN_End_Node example from ST severely breaks if you attempt to change the configuration with Cube.

Pillar1989, 2021-12-09:
@danak6jq To put it simply, we just update to project to v1.1.0. and make this project unrelated to CubeMX ?

danak6jq, 2021-12-09:
@Pillar1989 I would suggest looking at this repo: https://github.com/danak6jq/Seeed-LoRa-E5 . 

This is a _newly-created_ project not derived from the ST example. Instead, this uses Cube's configurator to populate the LoRaWAN application framework. Clone this repo, use "Open Projects From Filesystem" and select the cloned repo. I encourage you to clone the repo and give it a try.

After creating this project, I merged the source of lora_app.c from the ST LoRaWAN_End_Node application with that in my project so it has the same functionality.

[image]

Then you build the project, and, as desired, run Cube to edit Seeed-LoRa-E5.ioc to change the project configuration, generate code and reports

[image]

It is effectively an update of the example application, but it's not done by patching files in ST's example, it's a different project.

Pillar1989, 2021-12-11:
https://github.com/Seeed-Studio/LoRaWan-E5-Node/tree/qian 
@danak6jq can you give a glance at this branch? We want repository examples to support cleanly regenerates and builds, and Multiple examples can be supported simultaneously.

danak6jq, 2021-12-13:
@Pillar1989 is this project still built by copying files into the SDK example code?

[image]

That's not the correct way to do this. If you look at my repo, I generate the project using CubeMX without modifying the SDK example directly.

lakshanthad, 2022-03-08:
Hello @danak6jq,

Could you please have a look at the repo now? It has been updated.
https://github.com/Seeed-Studio/LoRaWan-E5-Node

Thank you.

descartes, 2022-03-08:
This has multiple issues, not least the apparent lack of knowledge of CubeMX.

The `LoRaWAN_End_Node.ioc` is a direct copy of the ST supplied file with the MCU changed from the variant used on the Nucleo board to the E5.

Consequently when you do something useful with the CubeMX, like configure some peripherals, in my case, I2C and some GPIO, the updated files barely start compiling before errors come up:

```
In file included from ../../LoRaWAN/Target/radio_conf.h:31,
                 from /Users/nick/TestingE5/LoRa-E5/SeeedMainNew/Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver/radio_driver.c:34:
../../Core/Inc/platform.h:46:10: fatal error: stm32wlxx_nucleo.h: No such file or directory
   46 | #include "stm32wlxx_nucleo.h"
      |          ^~~~~~~~~~~~~~~~~~~~
In file included from ../../LoRaWAN/Target/radio_conf.h:31,
                 from /Users/nick/TestingE5/LoRa-E5/SeeedMainNew/Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver/radio.c:39:
../../Core/Inc/platform.h:46:10: fatal error: stm32wlxx_nucleo.h: No such file or directory
   46 | #include "stm32wlxx_nucleo.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
make: *** [Middlewares/SubGHz_Phy/subdir.mk:25: Middlewares/SubGHz_Phy/radio.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Middlewares/SubGHz_Phy/subdir.mk:27: Middlewares/SubGHz_Phy/radio_driver.o] Error 1
In file included from ../../LoRaWAN/Target/radio_conf.h:31,
                 from /Users/nick/TestingE5/LoRa-E5/SeeedMainNew/Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver/radio_fw.c:26:
../../Core/Inc/platform.h:46:10: fatal error: stm32wlxx_nucleo.h: No such file or directory
   46 | #include "stm32wlxx_nucleo.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Middlewares/SubGHz_Phy/subdir.mk:29: Middlewares/SubGHz_Phy/radio_fw.o] Error 1
"make -j5 all" terminated with exit code 2. Build might be incomplete.
```

So this repro is the ST`LoRaWAN_End_Node` that has been changed to support the E5 with some other apps added in, but not setup to support the E5 mini or dev board. This is made rather obvious if you look at the GPIO that is for the Nucleo board and bears no resemblance to the pin table in the main ReadMe.

And it was clearly not tested - but then I don't think how CubeMX helps for quick prototyping or supporting something complicated like the LoRaMAC-node is understood here so I guess it shouldn't be a surprise.

I'd pull the .ioc files in this repro to stop people running in to problems and point people to the @danak6jq repro which works exactly as CubeMX is meant to.

danak6jq, 2022-03-08:
I echo @descartes comments. It's not using CubeMX as intended. Have a look at the repo I published, which doesn't use the example code at all, but is a new, independent CubeMX project. Feel free to clone my repo and start from there.

zhanggw7, 2022-04-08:
Hi @danak6jq 
I has build the project and the project has sucess wrok, I think CubeMX could be better as you said, I will try it later, thanks for your suggestion .

Best reagards

thonghatmit, 2024-05-24:
> I echo @descartes comments. It's not using CubeMX as intended. Have a look at the repo I published, which doesn't use the example code at all, but is a new, independent CubeMX project. Feel free to clone my repo and start from there.

Hi @danak6jq 
I have use your repo but the adc pin can not be generated, it break down the code like in this repo

mingzhangqun, 2024-10-16:
@jpmeijers I'm interested if your problem has been solved.

HeadBoffin, 2024-10-16:
@mingzhangqun, this is NOT how GitHub works - asking on every issue will end up with my mailbox and other peoples mailbox with notifications where you are asking if some ancient issue is solved when you could determine that either by age or by content.

Also, adding tags to every issue of `LoRaWan-E5-Node` in the LoRaWan-E5-Node repro is just spam - if the issue isn't about the repro, just close it.

closed 2024-10-17 by mingzhangqun