Seeed-Studio/LoRaWan-E5-Node #4: Error: Unable to get core ID at step 12

Julie97434, 2022-04-20:
Hello, 
I followed all the steps to the letter, but I have an error when I execute step 12. I want to program the LoRa E5 mini.

At step 11: I click on the connect button in STM32CubeProgrammer and get the following information:
[image]

For step 12: I select my .hex file and as soon as I click on `Start Programming` I get the error: `Error: Unable to get core ID` and the board disconnects from STM32Cube pro. As in the following photo: 
[image]

What does that mean ? Does this have anything to do with the devEUI I write in the `se-identity.h` file?

lakshanthad, 2022-04-28:
Hello @Julie97434,

Have you set the "Read Out Protection" to AA?

Best Regards,
Lakshantha

Julie97434, 2022-04-29:
Hello @lakshanthad , thank you for your answer. 

I misread the information, it's the ST-Link card (nucleoL073RZ) that disconnects after the successful upload of the program in the LoRaE5-mini. So everything works !

lakshanthad, 2022-04-29:
That's great!

closed 2022-04-29 by lakshanthad