Seeed-Studio/LoRaWan-E5-Node #5: "Multiple definitions of hlptim1" when running FreeRTOS project

ES0R, 2022-05-03:
I completed the LoraWAN end node tutorial for the LoRA E5 Dev board (https://wiki.seeedstudio.com/LoRa_E5_Dev_Board/#freertos-lorawan) and continued to the RTOS part. However, when running the code unaltered, it fails to compile. This is the case for both FreeRTOS and FreeRTOS_AT. I have been unable to locate the error. I have attached the console log for when it fails.

[log.txt](https://github.com/Seeed-Studio/LoRaWan-E5-Node/files/8610726/log.txt)

rpiper, 2022-05-08:
I just ran into this same issue when building the different FreeRTOS examples. There is a simple fix.

For each project, in Core/Inc/main.h, find the line:

`LPTIM_HandleTypeDef hlptim1;`

and change it to:

`extern LPTIM_HandleTypeDef hlptim1;`

and all three projects will build without any errors or warnings!

rpiper, 2022-05-08:
#6

lakshanthad, 2022-05-17:
It is done.
https://github.com/Seeed-Studio/LoRaWan-E5-Node/pull/6

closed 2022-05-17 by lakshanthad