Seeed-Studio/LoRaWan-E5-Node #7: FreeRTOS LoRaWAN AT Command issues

ES0R, 2022-05-19:
When running the given AT commands from the FREERTOS LoRaWAN AT build some of the commands fail. 
I was able to connect to TTN using "AT+JOIN=1" and I was able to use "AT?" but using other commands seems to just result in "AT_ERROR". For example, using "AT+VER" gives an error.  I have been unable to locate the error, can this be replicated by any others?

lakshanthad, 2022-05-20:
Hello,

Please help me answer the following:

- What is the exact E5 product you are using?
- What serial terminal application are you using to connect with LoRa E5?
- What are the connection parameters such as the baud rate you have set up?
- Did you add '\r\n' at the end of the command?

Thank you

ES0R, 2022-05-25:
Hello

Sorry for late answer: 
I'm using the LoRa-E5 Dev Board. 
I'm using arduino's serial monitor
I'm using 115200 as baudrate
Yes

lakshanthad, 2022-05-29:
Hello,

I will test this and get back to you.

lakshanthad, 2022-05-30:
Hello,

I have tested this and you should enter the commands as follows:

To get a value: 
`AT+<CMD>=?`

[image]

Therefore to get the firmware version, you should enter:
`AT+VER=?`

Make sure of these settings also:
[image]

Please let me know how it goes.

Thank you.

shawnventi, 2022-08-03:
I believe that some confusion in this issue might be from the fact that the factory AT firmware has a different AT command spec than the project included in this repo.

The factory AT firmware follows this spec: https://files.seeedstudio.com/products/317990687/res/LoRa-E5%20AT%20Command%20Specification_V1.0%20.pdf

Is it possible to get a copy of the factory AT firmware that can be used?

closed 2022-08-05 by ES0R

bkrajendra, 2023-07-01:
Hi, I am also trying to work with FreeRTOS AT firmware.
I can get a response for AT?, AT+VER=? etc. But whenever I try to set the value, I get Error AT_PARAM_ERROR
What is the right format to set value with AT.
I tried AT+DADDR=xxxx, but with no success.