Seeed-Studio/LoRaWan-E5-Node #8: IOC Faliure

e135193, 2022-05-19:
FreeRTOS projects works well but there is a problem with the "FreeRTOS_LoRaWAN.ioc" file. When I open the IOC file, configuration is not showing correctly. It seems like it is a fresh project. Could you please check and correct it.

In addition, I can not debug the End Node project. stlink loses connection with the MCU.

[image]

lakshanthad, 2022-05-30:
Hello,

I will check about the IOC file.

In the meantime, could you please elaborate more on the STLink disconnection?

1, Does it connect successfully for the first time?
2, After how long it gets disconnected?
3, Can I see a photo of the connection?
4, Have you set "Read Out Protection" as AA in "STM32CubeProgrammer"?

Thank you.

sjackson2011, 2022-11-11:
Any answer to the original .ioc file problem? It seems that the project has not been built with the .ioc file included. Specifically I am using the w5_mini  FreeRTOS_LoRaWAN_AT project

mingzhangqun, 2024-10-16:
I'm interested if your problem has been solved.

closed 2024-11-04 by mingzhangqun