Seeed-Studio/LoRaWan-E5-Node #9: Simple LED blink on lora-e5 mini

kargarisaac, 2022-07-01:
Hello,

I'm trying to use this repo to program a new lora-e5 mini board that I got recently, but cannot program the board. So, I decided to just test a simple LED blink test. Again, I cannot do it and get the same error. This is the error I get:

```bash
STMicroelectronics ST-LINK GDB server. Version 6.1.0
Copyright (c) 2022, STMicroelectronics. All rights reserved.

Starting server with the following options:
        Persistent Mode            : Disabled
        Logging Level              : 1
        Listen Port Number         : 61234
        Status Refresh Delay       : 15s
        Verbose Mode               : Disabled
        SWD Debug                  : Enabled
        InitWhile                  : Enabled

Waiting for debugger connection...
Debugger connected
Waiting for debugger connection...
Failed to read registers from target
Shutting down...
Exit.
```

I create a new empty project and set portB pin5 on and off as follows:

```C
while (1)
  {
	  HAL_GPIO_WritePin(GPIOB, GPIO_PIN_5, GPIO_PIN_SET);
	  HAL_Delay(1000);
	  HAL_GPIO_WritePin(GPIOB, GPIO_PIN_5, GPIO_PIN_RESET);
	  HAL_Delay(1000);
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
```

Any idea? 

I also tried to use stm32cubeProgrammer but it didn't help:

```bash
  22:33:11 : Memory Programming ...
  22:33:11 : Opening and parsing file: loraE5Mini_blink.elf
  22:33:11 :   File          : loraE5Mini_blink.elf
  22:33:11 :   Size          : 6.20 KB 
  22:33:11 :   Address       : 0x08000000 
  22:33:11 : Erasing memory corresponding to segment 0:
  22:33:11 : Erasing internal memory sectors [0 3]
  22:33:11 : Download in Progress:
  22:33:11 : Error: failed to download Segment[0]
  22:33:11 : Error: failed to download the File
  22:33:11 : RUNNING Program ... 
  22:33:11 :   Address:      : 0x08000000
  22:33:11 : Warning: The core is locked up
  22:33:11 : Start operation achieved successfully
```

lakshanthad, 2022-07-04:
Hello @kargarisaac,

Have you set the "Read Out Protection" to AA as mentioned in the README?

Best Regards,
Lakshantha

johflo, 2022-07-19:
Hi,

Change RDP to AA and press Apply.
This works for me.

Best Regards,
Johannes

mingzhangqun, 2024-10-16:
I'm interested if your problem has been solved.

mingzhangqun, 2024-11-04:
Thanks for your feedback.