Seeed-Studio/LoRaWan-E5-Node #10: Timout error in SUBGHZ_SPI_TRANSMIT function

hpernet, 2022-07-13:
Hello, 

I'm trying to developpe a P2P application between two LoRa-E5 mini. 
I reprogram the STM32WL on the board using STM32CubeIDE.

At first, I try to developpe my own application from PINGPONG exemple of STMicroelectronics and your exemples, but while debuging, I found that, in SUBGHZ layer, the SUBGHZ_SPI_TRANSMIT function return Timout error, its looks like the TXE flag of the SubGHz is never set...

I first think that it was from my code, but then, I try to use your LORAWAN end node code exemple, and fall into the same error. The only thing that I modify is that I comment the debug initialization function to be able to debug the code.
I also ty to disable Low Power manager to see if there is a conflict between debug and LPM but nothing change.

Do you have an idea about whats happenning ? 
Do you think this may come by Option byte configuration, or debug configuration ?

You can find here the only things that I try to modify in sys_app.c file : 
[image]

And my debug config :

[image]

johflo, 2022-07-29:
Hi,
I made some changes on the STM PingPong example (configuration based on (https://github.com/danak6jq/Seeed-LoRa-E5) ) without errors.
Perhaps you can find the difference of your example.

Johannes

hpernet, 2022-07-29:
Hi @johflo, 

Thanks so much for your answer ! 

I tried your project and I've got the same error... 
It's must come from my side but I don't know whitch configuration...

Do you have any radio problem while using the debug ? 
Is it possible for you to send me your option byte configuration ? 
Did you make some changes on the board to make it work ? 

Thanks 
Hugo

johflo, 2022-07-29:
Hi,

I tested the programm without errors.
since this is my first program I can only debug the init part. After the init the program/debugging changed to run mode so I can't debug the hole program.
In the init part I didn't get some errors.

Johannes

hpernet, 2022-07-29:
Okey thanks ! 

My error occured in the initialization of the radio.
In stm32wlxx_hal_subghz.c file, when SUBGHZSPI_Transmit function is called.

[image]

Maybe that the radio cannot works while debugging...

For option bytes they can be found using STM32Cube debugger in OB menu when you are connected to your device. You should have modify them to program the board with STM32CubeIDE I think.

Hugo

johflo, 2022-07-29:
I will take a look in the evening and I'll report you the settings

johflo, 2022-07-29:
Hi,
I've uploaded the code again. There were some mistakes from the first upload. now should it work.

Johannes

hpernet, 2022-08-01:
Hi @johflo, 

Thank you very much for your time ! 

But I still have the same problem... 

I even try to program the board using STM32CubeProgrammer and the .hex file, and to put en NUCLEOSTM32WL with pingpong code, but no communications seams to happens. 
I don't know whats happenning... 
It's don't seams to be a firmware problem since I'm using yours.

Did you succed to reproduce my error or is it only on my side ? 
Did you succeed to make some transmissions ? 

Hugo

johflo, 2022-08-01:
Hi,

I can't reproduce your error.
I have two LoRa-e5 mini with wich I can run the program normally. One sends the "Ping", the other sends the "Pong".

I also debugged the programm but I didnt get your error.
Do you tested with my Hex-File?

Johannes

johflo, 2022-08-01:
What type of debugger do you use? STLink V2?

hpernet, 2022-08-01:
I tested with your hex-file, and two LoRa-e5 mini, but I still have the error.
I open a serial interface for debug, and I have only Timout errors : 
`PING PONG
APPLICATION_VERSION: V1.2.0
MW_RADIO_VERSION:    V1.2.0
---------------
LORA_MODULATION
LORA_BW=125 kHz
LORA_SF=7
36s278:rand=0
41s005:OnRxTimeout
43s340:Master Tx start
64s499:OnTxTimeout
66s630:Slave Rx start
71s361:OnRxTimeout
73s697:Master Tx start
94s859:OnTxTimeout
96s992:Slave Rx start
101s723:OnRxTimeout
`

I use a STlink V2 from a Nucleo board for debug.
Do you use thepower source of the debuger or of the USB connector ?

johflo, 2022-08-01:
on debugg mode i use only the STLink.
On terminal mode (2 arduino serial monitors) I use USB power.

johflo, 2022-08-01:
On the second device do you have the same output?

hpernet, 2022-08-01:
Do you have this type of stlink ? 
https://www.amazon.fr/ElectroWorldFR-Simulateur-Programmeur-Downloader-D%C3%A9bogueur/dp/B09HJQQ1C6/ref=sr_1_1_sspa?keywords=stlink+v2&qid=1659342438&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyOElTUUZENTNQVlMmZW5jcnlwdGVkSWQ9QTA2OTQwNzIxQzcyOFRBSVowOE5aJmVuY3J5cHRlZEFkSWQ9QTAyOTc0NDkxMzhRRUtCMFBTTVMxJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

Yes I have the same output on both devices

hpernet, 2022-08-01:
On debbug mode, do you power the device with 3.3V or 5V ?

johflo, 2022-08-01:
I have the same STLink V2. I updated it with the last firmware.
I power the devices with 5V.

do you have on both devices the rand=0?

hpernet, 2022-08-01:
I finally succeed !! 

I had a third board that I never used before, so I tried to flash it and it works ! 

I don't know whats happenning with both others boards...

Yes I have rand = 0 on both devices. I think that is because rand use a subghz feature and the mcu can't communicate with the subghz modem.

johflo, 2022-08-01:
Hi, 
Good to hear.
I have on both devices randoms different of 0.
One or two times I also have strange behaviours (no debugger found). After a erase in the STM32CubeProgrammer the problem was solved.

Advanced-Machine-Intelligence-AMI, 2024-10-04:
can i get your hex file i am facing issue where rand = 0. Also please tell me whether i should flash same code to both boards without any changes.

johflo, 2024-10-04:
You can Download the Code and compile it. you can flash reg Output on Bothe devices without changes

mingzhangqun, 2024-10-16:
Hello, I'm interested if your problem has been solved.

closed 2024-11-04 by Lesords