Seeed-Studio/LoRaWan-E5-Node #11: Clock issues on custom board

Kakarotas, 2022-09-05:
- Examples are running super slow, it takes 33 seconds to reach "TX on freq 902500000 Hz at DR 0" phase. I believe clocks are wrong frequency:

`###### OTAA ######
###### AppKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### ABP  ######
###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEui:  2C:F7:F1:20:XX:XX:XX:XX
###### AppEui:  80:00:00:00:00:00:00:06
###### DevAddr: 00:17:F6:44
33s019:TX on freq 902500000 Hz at DR 0
###### OTAA ######
###### AppKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### ABP  ######`
And after another 10-20 seconds it seems to reboot since code output repeats. 
In seed studio wiki I can see that code should get there in like 36ms according to the [screenshot](https://files.seeedstudio.com/wiki/LoRa-E5-mini/TTN-joined.png). 

FreeRtos example it is 40 seconds to reach "TX on freq" and "Mac txTimeout" instead of reset. 

`----------LoRaWAN_End_Node_FreeRTOS----------

###### OTAA ######
###### AppKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### ABP  ######
###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEui:  2C:F7:F1:20:XX:XX:XX:XX
###### AppEui:  80:00:00:00:00:00:00:06
###### DevAddr: 00:17:F6:44
40s091:TX on freq 902500000 Hz at DR 0
65s682:MAC txTimeOut

###### = JOIN FAILED
71s683:temp= 27
79s790:TX on freq 905300000 Hz at DR 0
101s397:temp= 27
103s272:temp= 27
105s396:MAC txTimeOut`

Details:
Trying to run end node examples on my custom board using the modules bought from Digikey. Currently custom board is populated only with power supply and E5 module with a single led on PC0 (not used in example projects as far as I can see) and one power supply sleep control pin which is input (PB3). RST is connected to a button like in dev kit. I set RDP to AA as instructed, hex is flashing fine. Same happening even if I won't modify DevEui and region, only adding Hex generation for post build output on the clean project. STM32CubeIDE (Version: 1.10.1) and STM32CubeProgrammer(v2.11.0) are both freshly installed. Board is running on 3.3V, tried to run it from external power supply without any changes too, it consumes few mA max, module is bypassed with 10uF and 4.7uF caps. 

Any tips would be appreciated. Thanks.

PeterHRO, 2023-01-06:
I had the same problem. And it may be that the modules broke for some reason. I had previously tested the modules with the AT software, then deleted and reprogrammed them. I had the same result. It was very slow end it took 33s to reach the message. Then came a MAC TX_Timeout. 
As a programmer I had a cheap USB stick from Amazon. For the third module I used a different programmer from an old Discovery board. Then it worked.

mingzhangqun, 2024-10-16:
Hello, I'm interested if your problem has been solved.

closed 2024-11-04 by Lesords