Seeed-Studio/LoRaWan-E5-Node #12: Compile error after changing .ioc

pablo-1424, 2022-10-28:
Hello, 

My goal is to work with IrDA on the UART1 and once I've generated the code from the .ioc, there 12 compilation errors (see attachement 1).
Have someone had the same issue ? I have attached the modified UART in the .ioc (see attachement 2).

Thank you,

Pablo
[attachement1.txt](https://github.com/Seeed-Studio/LoRaWan-E5-Node/files/9886809/attachement1.txt)
[image]

thonghatmit, 2024-05-24:
i have same issue, maybe code from repo does not lying in user section tag, so it break down the code when you generate ioc file. Do you have some solution ?

mingzhangqun, 2024-10-16:
I'm interested if your problem has been solved.

closed 2024-11-04 by Lesords