Seeed-Studio/LoRaWan-E5-Node #14: Help - Factory firmware -  E5's project folder

ShanmugaCA, 2023-03-06:
Hi!

All the E5 modules are programmed by Seeed Studio and I want to do some changes to that firmware (FW)  and use it!

So, can anyone help me to locate the FW project folder of E5 module ?

Many thanks in advance,

Best regards,
Shanmuganathan

sm-thamizha, 2023-03-24:
The factory firmware isn't available for public. You're either stuck with using the E5 modules as LoRa nodes with the factory firmware or you can reprogram the board entirely and use it for your application.

ShanmugaCA, 2023-03-24:
It is unfortunate that the firmware of the E5 module is not available for build.

bkrajendra, 2023-07-01:
HI @ShanmugaCA 
Although factory AT firware is not opensource but there is a FreeRTOS AT source under folder Projects/Applications/FreeRTOS/FreeRTOS_LoRaWAN_AT
I am also working on it but could not get it working. Most if the AT command does not work. Might be I am missing something.

ShanmugaCA, 2023-07-07:
Hi@bkrajendra,
Thanks for reply!
BTW, which Country specifi frequency you are trying?
I am intrested in making this work for India specific range and want to use the controller inside the chip than an external one.

bkrajendra, 2023-07-08:
Update:
Projects/Applications/FreeRTOS/FreeRTOS_LoRaWAN_AT worked for me. The issue was the format of AT commands. 
AT+<CMD>=Value
this works. I was trying for IN865.
Find some notes:
**How to do OTAA**
In AT command source code change lorawan_conf.h file to uncomment **IN865** region

In Arduino Serial terminal set to 115200 + Carriage Return
**AT commands to use:**
Change Region: AT+BAND=7
Join OTAA: AT+JOIN=1
Send : AT+SEND=1:1:123

bkrajendra, 2023-07-08:
I am also trying to use the internal STM32 for creating a custom sensor node.
FYI, there is an Arduino support for this https://github.com/stm32duino/Arduino_Core_STM32
And a library which I am also working on to make it work. https://github.com/stm32duino/STM32LoRaWAN
check these out.

mingzhangqun, 2024-10-16:
@ShanmugaCA @bkrajendra  Hello, I'm interested if your problem has been solved.

closed 2024-11-04 by Lesords