Seeed-Studio/LoRaWan-E5-Node #15: FreeRTOS examples do not compile under STM32CubeIDE v 1.12

jkuzma, 2023-05-23:
The FreeRTOS examples in this repository do not compile properly with the latest version of STM32CubeIDE.

mingzhangqun, 2024-10-16:
@jkuzma Hello, I'm interested if your problem has been solved.

Lesords, 2024-11-04:
Hello,

I\'m going to close this issue, feel free to re-open it if you have any other questions.

[Autocommentary]

cristinel08, 2024-11-07:
Hello,
I keep getting multiple definition error of hlptim1 when trying to compile FreeRTOS_LoRaWAN_AT. 
I am using STM32CubeIDE v1.16

Lesords, 2024-11-07:
> Hello, I keep getting multiple definition error of hlptim1 when trying to compile FreeRTOS_LoRaWAN_AT. I am using STM32CubeIDE v1.16

@mingzhangqun help with this one

mingzhangqun, 2024-11-08:
@cristinel08 Hello, I'm reproducing the problem.

closed 2024-11-28 by mingzhangqun