Seeed-Studio/LoRaWan-E5-Node #16: no STM32 target after programming the board

Sebelectronic, 2023-08-29:
Hi, I have followed your instructions until the end, compile the soft without problem , program it fine. But now when I want to program the LORA-E5 Mini again the board isn't detected by the programmer. I've tried pressing reset and boot push button without results.
Any idea to help me to save my board?
Thank you

LucasAMPA, 2023-08-29:
Jut to get an idea of what your set up is:

1. How are you flashing the board? USB, SWD, or something else?
2. What software are you using to flash the board? What are the program names/IDE names specifically?
3. What machine are you using, Linux, Windows, or OS?

An easy mistake is to remove the factory given AT from Seeed by flashing over it, as an FYI.

Sebelectronic, 2023-08-31:
I flash the board with a STLINK-V2 clone
I am under Ubuntu18.04 with STM32CubeIDE and program with STM32Programmer

To give more details, my board was with AT firmware and first I tried flashing the "official" STM32 lorawan stack(over it). And I could do this several times. successfully Then I decided to use the source from here with is dedicated to LORA-E5 mini and I worked fine but only once :-) 
I am a newbie with STM32 architecture...

miroslavpetrov, 2023-10-14:
@Sebelectronic did you fix the issue? I am experiencing the same issue now. I have followed the tutorial and flashed the lora-e5 with the LoRaWAN_End_Node firmware. Now the board cannot connect to stm32cubeprogrammer... I use v3-minie programmer.

Sebelectronic, 2024-07-03:
@miroslavpetrov , sorry for my late answer but I didn't try to fix my trouble. Did you fix it?

miroslavpetrov, 2024-07-03:
@Sebelectronic I use genuine STLink V3-Minie. It turned out that this little sh** works 100% of the time with only one of 10+ tested USB-C to USB-C cable and it works fine ONLY if the cable is plugged in a certain way(if I reverse it it doesn't work).

mingzhangqun, 2024-10-16:
@Sebelectronic Hello, I'm interested if your problem has been solved.

closed 2024-11-04 by mingzhangqun