Seeed-Studio/LoRaWan-E5-Node #20: E5-mini STM32CubeIDE LORAWAN_End_Node wrong IOC File

wpfundstein, 2024-02-27:
Hi,

i downloaded the LORAWAN_End_Node Project for STM32CubeIDE

i can compile and upload the hex file to the mini-E5 and it works.

Now i want to configure some Input and Output GPIO but when i save and generate the code i get errors. the reason is incomplete configuratation in LORAWAN.

i think the ioc file is not for the mini-E5

what is wrong?

thanks, Werner

marekpow, 2024-02-27:
Try this: [link](https://github.com/Seeed-Studio/LoRaWan-E5-Node). It works for me.

wpfundstein, 2024-02-27:
> Try this: [link](https://github.com/Seeed-Studio/LoRaWan-E5-Node). It works for me.

i work with this project files and have issues with the ioc file.

thanks

wpfundstein, 2024-03-20:
No. If i Change anything. Add a gpio or other and then create Code i get many Errors by compiling the Code.

marekpow, 2024-03-20:
Try to use [my](https://github.com/marekpow/Wio5E-mini). I it forked from the above repo.

wpfundstein, 2024-03-20:
Try to use [my](https://github.com/marekpow/Wio5E-mini). I it forked from the above repo.

your link is not working.

marekpow, 2024-03-25:
I'm sorry, it was my mistake. The proper URL is [here](https://github.com/marekpow/lorawan-test). 
I forked it from the Seeds Studio repo via a zipped code.

thonghatmit, 2024-05-23:
> I'm sorry, it was my mistake. The proper URL is [here](https://github.com/marekpow/lorawan-test). I forked it from the Seeds Studio repo via a zipped code.

Hi
Sorry, but  i use your repo on the link, the problem is still there when i try assign adc pin for ioc file, when code is generated, it get the same error, file adc_if.c is auto deleted by code generator
[image]

mingzhangqun, 2024-10-16:
Okay, I'll repeat your steps and find out why.

closed 2024-11-04 by Lesords