Seeed-Studio/LoRaWan-E5-Node #24: How to setup multicast transmission with Chirpstack

be-mar, 2024-11-11:
Hi all,
I want to establish multicast communication between chirpstack server and WIO-e5 mini modules.

Currently as a test base I'm using Lorawan-end-node. Unicast communication in class C works perfectly but seems like multicast messagess doesn't reach my debug module.

I've debugger connected to WIO-e5 development kit. Each time I try to send message from chirpstack using swagger API debugger doesn'y stop in defined breakpoint. I assume there is a problem with keys.

Could someone give me a guidence how to setup proper multicast communication?
Thanks

mingzhangqun, 2024-11-13:
I would like to confirm with you whether it is one node multicast transmission to multiple chirpstacks? Did you correctly add nodes to each chirpstack? Can you provide your detailed configuration instructions?

be-mar, 2024-11-14:
Sorry for late reply. 
One chirpstack server - to two C class devices.

Not sure what you are asking about but here:

Gateway works as a packet forwarder:
[image]

Gateway is defined in chirpstack:
[image]

Unicast frames are visible here:
[image]
 Thing is that when I use api to post a multicast message the message itself is not visible in that log...

Device profile defined here:
[image]

Class C support enabled
[image]

Devices attached to application:
[image]

Mulitcast group with defined devices:
[image]

And gateway:
[image]

Config of multicast group:
[image]

Unicast messages from end device:
[image]
[image]

mingzhangqun, 2024-11-15:
Make sure your module has multicast enabled.
https://files.seeedstudio.com/products/317990687/res/LoRa-E5%20AT%20Command%20Specification_V1.0%20.pdf
[image]

be-mar, 2024-11-15:
I tried to use this function for stock chip by typing at commands. Nothing happened. Then I decided to build [lora-end-node](https://github.com/Seeed-Studio/LoRaWan-E5-Node/tree/main/Projects/Applications/LoRaWAN/LoRaWAN_End_Node) project and manually define keys in se-identity.h but same. I can set breakpoints and see only unicast messages

be-mar, 2024-11-16:
BTW. I use OTAA activation. Does it matter?

mingzhangqun, 2024-11-18:
> I tried to use this function for stock chip by typing at commands. Nothing happened. Then I decided to build [...]

Can you tell me all your commands?

be-mar, 2024-11-27:
Have you had an opportunity to check this issue?

mingzhangqun, 2024-11-29:
Sorry, I am dealing with urgent work this week, I am ready to debug today.
I think the problem should be to open MC in lora-end-node, I will try it next week.

mingzhangqun, 2024-12-03:
`AT+ CLASS=C` must run after `AT+join`. Can you give it a try and let me know?
Thank you.

mingzhangqun, 2024-12-03:
btw, which version of chirpstack do you use?

mingzhangqun, 2024-12-04:
I've tried on two devices E5 nodes COM77 and COM78. 
COM77
» AT+ID
« +ID: DevAddr, 01:56:81:C9
+ID: DevEui, 2C:F7:F1:20:24:40:00:EA
+ID: AppEui, 80:00:00:00:00:00:00:06
AT+MODE=LWOTAA
AT+DR=EU868
AT+ID=DevAddr,"015681c9"
AT+KEY=APPKEY,"a98dd0ae84f9778f7720ca844785b04e"
AT+KEY=NWKSKEY,"650b74915ead6846345de75508fb5b50"
AT+KEY=APPSKEY,"7bb1d5c7fc950a66c3c8576129ddef12"
AT+LW=MC,ON,"019510a0","850f4525bb585e6ed9afa2fa72dc63d0","e28737a5ac72d22dce22c0b299db2073",0
AT+JOIN
AT+CLASS=C

COM78
» AT+ID
« +ID: DevAddr, 01:1E:96:A0
+ID: DevEui, 2C:F7:F1:20:32:30:C4:9B
+ID: AppEui, 80:00:00:00:00:00:00:06
AT+MODE=LWOTAA
AT+DR=EU868
AT+ID=DevAddr,"011e96a0"
AT+KEY=APPKEY,"2b691f7af4c6f332c98e144085f8607d"
AT+KEY=NWKSKEY,"64726033f21b8369b4fc3508ca9b2b40"
AT+KEY=APPSKEY,"3f330c7777a917f26e6bee74f8a20a2a"
AT+LW=MC,ON,"019510a0","850f4525bb585e6ed9afa2fa72dc63d0","e28737a5ac72d22dce22c0b299db2073",0
AT+JOIN
AT+CLASS=C

[image]
[image]

[image]
[image]
[image]

be-mar, 2024-12-04:
AD [2513928451](https://github.com/Seeed-Studio/LoRaWan-E5-Node/issues/24#issuecomment-2513928451)
I tested it previously on Chirpstack v3 and v4 with the same results. I'll try that order of commands.

Where can I find these values?
_AT+KEY=APPKEY,"a98dd0ae84f9778f7720ca844785b04e"
AT+KEY=NWKSKEY,"650b74915ead6846345de75508fb5b50"
AT+KEY=APPSKEY,"7bb1d5c7fc950a66c3c8576129ddef12"_

be-mar, 2024-12-04:
Update. Tested with AT commands and works! But as you can see I didn't specify AT+KEY=APPKEY, AT+KEY=NWKSKEY, AT+KEY=APPSKEY.
[image]

Have to verify if will that work with lora-end-node project. Could you give me some tips what needs to be modified there to enable multicast messages?

mingzhangqun, 2024-12-05:
> AD [2513928451](https://github.com/Seeed-Studio/LoRaWan-E5-Node/issues/24#issuecomment-2513928451) I tested it previously on Chirpstack v3 and v4 with the same results. I'll try that order of [...]

These keys are randomly generated.

mingzhangqun, 2024-12-05:
Currently the lora-end-node project is not verified with multicast.
You can reference this repo:
https://github.com/SylvainMontagny/STM32WL

Thank you.

be-mar, 2024-12-05:
_ These keys are randomly generated._ So should I skip this step or generate this keys and somhow store them in server. Sorry, I don't get it 🤔

mingzhangqun, 2024-12-05:
Save both in server and e5-node.

be-mar, 2024-12-12:
I'm closing this issue as it is working with at commands. Thanks

be-mar, 2025-02-12:
@mingzhangqun one side question. How did you set chirpstack to see that Queue tab in MC group?

It's apparently missing from my server..

[image]

mingzhangqun, 2025-02-17:
@be-mar There may be differences between versions, please try v4.10.1.
Thank you.

be-mar, 2025-02-17:
Yep, I had v4.9. After upgrade to 4.11 I can see this tab. Thx

closed 2025-02-17 by be-mar