Seeed-Studio/LoRaWan-E5-Node #25: Codebase huge changes after adding one adc pin in .ioc and regenerating

wql2002, 2024-11-27:
Hi,

Thanks for all your effort in example code and documentation! I am using a Wio Lora-e5 dev kit board and successfully run the `LoRaWAN_End_Node` example to send messages to TTS server.

Now I want to customize this example by adding one adc pin input, so I edited the .ioc file and added one pin. However after regenerating the project, I found that the codebase had huge changes and could not compile.

So my current solution is to manually add some init code for the pin in the original adc.c instead of using .ioc to regenerate the project. But this solution is apparently not scalable or flexible.

I will appreciate any suggestions on how to develop own applications based on this example.

Thanks,
Qianliang

mingzhangqun, 2024-11-29:
Can you show me how you added the ADC (which pin), and your cube ide version?

wql2002, 2024-11-29:
I added the PB3 -> ADC_IN2, and my cube ide version is `MxCube.Version=6.8.0`.

Because I am a newbie in stm32 programming, my solution might be the best practice:

1. used the original .ioc file to generate the code, which still caused huge codebase changes.
2. then added the PB3 pin in .ioc setting, and regenerated the code.
3. compared two versions of code, then manually applied the incremental change to a new clone of this repo.

The incremental change is actually very simple:

[image]

And it works!

But it is not very elegant and I want to know whether there are some best practices for future development.

Thanks,
Qianliang

mingzhangqun, 2024-12-04:
Yes, your point is right. This problem is caused by different versions, it is recommended that you use STM32cube ide v1.8.0. We will upgrade to support the latest version.

sambhavjain12, 2024-12-19:
We are facing the same issue. As you mentioned, we are unable to find STM32CUBE IDE version v1.8.0 in STM32 website

[image]

Can you share use the steps on how exactly to overcome this issue. 

Thanks in advance

mingzhangqun, 2024-12-26:
@sambhavjain12 Hello, you can compile this project directly follow the readme.md. It already includes the v1.8.0 sdk.

closed 2025-01-02 by mingzhangqun