Seeed-Studio/LoRaWan-E5-Node #27: Rx Timeout seen when joining lorawan using chirpstack

umamaheshrouthu, 2025-01-22:
I am using the code from github repo.
but lorawan joining is failing with below error.
the behaviour is same with all the configurations in this github repo

any idea how to fix it.
6:36:49.052 -> ----------LoRaWAN_End_Node_FreeRTOS----------
16:36:49.052 -> 
16:36:49.052 -> 
16:36:49.090 -> ###### OTAA ######
16:36:49.090 -> ###### AppKey:      30:78:01:EF:75:7E:E1:C0:E1:46:ED:8B:B8:70:46:27
16:36:49.090 -> ###### NwkKey:      D0:AF:BB:C5:C4:AB:98:42:83:8A:90:03:CB:FC:79:89
16:36:49.090 -> ###### ABP  ######
16:36:49.090 -> ###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
16:36:49.090 -> ###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
16:36:49.090 -> ###### DevEui:  70:B3:D5:7E:D0:06:D2:7F
16:36:49.090 -> ###### AppEui:  FF:EE:DD:CC:BB:AA:99:00
16:36:49.126 -> ###### DevAddr: 06:09:BB:9B
16:36:49.126 -> 0s039:TX on freq 868300000 Hz at DR 0
16:36:50.568 -> 1s526:MAC txDone
16:36:55.651 -> 6s561:RX_1 on freq 868300000 Hz at DR 0
16:36:57.429 -> 8s369:MAC rxDone
16:36:57.429 -> 
16:36:57.429 -> ###### = JOIN FAILED
16:36:57.429 -> 
16:36:57.429 -> ###### ========== MCPS-Indication ==========

marekpow, 2025-01-22:
Do you have a device attached to an application on the gateway on the Chirpstack using 
DevEui: 70:B3:D5:7E:D0:06:D2:7F
AppEui: FF:EE:DD:CC:BB:AA:99:00
AppSKey: 2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C

Do you have a device profile for his device on the Chirpstack side?

mingzhangqun, 2025-01-23:
Hi @umamaheshrouthu , You can refer to this configuration step:
https://github.com/Seeed-Studio/LoRaWan-E5-Node/issues/24

umamaheshrouthu, 2025-01-23:
> Do you have a device attached to an application on the gateway on the Chirpstack using DevEui: 70:B3:D5:7E:D0:06:D2:7F AppEui: FF:EE:DD:CC:BB:AA:99:00 AppSKey: 2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
> 
> Do you have a device profile for his device on the Chirpstack side?

Yes. i do have device profile, application and device as per these. in fact, with the same details, i see a lilygo tbeam able to join the lorawan network. so, there should be no issue with chirpstack configurations.

marekpow, 2025-01-24:
Check Lora REGION. Should be the same as for the gateway.

closed 2025-02-16 by github-actions[bot]