Seeed-Studio/LoRaWan-E5-Node #28: Unable to use AT commands other than AT? in FreeRTOS_LoRaWAN_AT

NNAA0917, 2025-04-17:
I have flashed the FreeRTOS_LoRaWAN_AT application in my environment, but I am unable to use any AT commands other than AT?. I receive an AT_ERROR response for all other commands.

Reproduction Environment:

SDK: STM32CubeIDE V1.17.0
Application: FreeRTOS_LoRaWAN_AT
Communication Software: Tera Term
Programming Tool: STM32CubeProgrammer
I built the application using STM32CubeIDE and flashed it using STM32CubeProgrammer. Upon startup, the communication software outputs the following logs:

APP_VERSION:        V1.1.0
MW_LORAWAN_VERSION: V2.3.0
MW_RADIO_VERSION:   V1.1.0

----------LoRaWAN_End_Node_FreeRTOS----------

###### OTAA ######
###### AppKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkKey:      2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### ABP  ######
###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEui:  00:80:E1:15:00:17:C7:22
###### AppEui:  01:01:01:01:01:01:01:01
###### DevAddr: 06:0E:FE:B1
ATtention command interface
AT? to list all available functions

After that, when I execute AT commands (e.g., AT+ID), I receive an AT_ERROR response. Additionally, I found that I need to set the baud rate to 115200 instead of 9600 for communication to work.

Is there any solution for this issue?

mingzhangqun, 2025-04-18:
Hi @NNAA0917 , this a stm32 official demo, the supported cmds are [here](https://github.com/Seeed-Studio/LoRaWan-E5-Node/blob/163c05379b1805dd8f2c061d4557a69985acc953/Projects/Applications/FreeRTOS/FreeRTOS_LoRaWAN_AT/LoRaWAN/App/lora_command.c#L97), there is no AT? or AT+ID cmd.
Thank you.

NNAA0917, 2025-04-18:
Hi @mingzhangqun ,
Thank you for your answer.
I understand now.
I also got confirmation of the command.
Thank you very much.

closed 2025-04-18 by NNAA0917