Seeed-Studio/LoRaWan-E5-Node #29: Request for Peer-to-Peer LoRa Communication Support in FreeRTOS_LoRaWAN_AT Application

NNAA0917, 2025-04-18:
I am currently using the FreeRTOS_LoRaWAN_AT application in my environment. While this sample application is primarily focused on connecting to LoRaWAN, I am looking to implement Peer-to-Peer communication using LoRa.

Specifically, I would like to know how to send data by specifying the channel and address in a simple manner.

Could you please provide guidance or suggestions on how to achieve this?

mingzhangqun, 2025-04-22:
Hi @NNAA0917 , You can use these [cmds](https://github.com/Seeed-Studio/LoRaWan-E5-Node/blob/163c05379b1805dd8f2c061d4557a69985acc953/Projects/Applications/FreeRTOS/FreeRTOS_LoRaWAN_AT/LoRaWAN/App/lora_command.c#L435C14-L435C24): AT+TCONF AT+TTX AT+TRX
This is the reference [document](https://www.st.com/resource/en/application_note/dm00346311-examples-of-at-commands-on-icubelrwan-stmicroelectronics.pdf).

Thank you.

NNAA0917, 2025-04-23:
Thank you for your confirmation. I have verified that I can communicate with the same LoRa-E5 Dev Board using the AT commands listed in the document. However, I am unable to send and receive data when trying to communicate with a different module (E220-900T225) using Private LoRa. Is it not possible to perform Private LoRa communication with a different module?

NNAA0917, 2025-04-24:
Thank you for your confirmation. The modules used in USB_E220-900T22S and Wio-E5 Development Kit are different (USB_E220-900T22S uses E220-900T22S, while Wio-E5 Development Kit uses SX126x). Is it possible to establish private LoRa communication between them?

mingzhangqun, 2025-04-25:
I think it's possible, be careful with the radio conf.
Thank you.

NNAA0917, 2025-04-25:
"Thank you for your confirmation. Understood. There is no problem closing this matter."

closed 2025-04-29 by mingzhangqun