Seeed-Studio/LoRaWan-E5-Node #30: MAC txTimeout

Nalan-S, 2025-04-19:
The serial monitor shows MAC txTimeout

[image]

Why does this happen?

mingzhangqun, 2025-04-22:
Hi @Nalan-S , have you added your node into the LoRaWan network Server(e.g., [Chirpstack ](https://github.com/Seeed-Studio/LoRaWan-E5-Node/issues/24))?

Nalan-S, 2025-04-22:
Yes, I have added it to The Things Network and was also able to connect it
using AT commands. I don't have those logs now.
While using this repo after erasing the AT firmware, I got the "MAC
txTimeout" response in the serial monitor which has been attached already.
Since the Lora-E5-mini couldn't send a join request I don't see any logs in
the network server.

marekpow, 2025-04-22:
Are you sure that you work with the 905 frequency?

Nalan-S, 2025-04-22:
Yeah! The gateway that I use works in the US915 frequency band. The same
has been configured in LoRa-E5-mini.

marekpow, 2025-04-22:
The UART console should show DevEUI, AppEUI, and JoinEUI. Did you check that the device profile you use has the same UEIs? Does your application use this configured device profile?

mingzhangqun, 2025-04-23:
@Nalan-S If you erased the  default AT firmware, the IDs will be lost. You need to set new params.
[cmds](https://github.com/Seeed-Studio/LoRaWan-E5-Node/blob/163c05379b1805dd8f2c061d4557a69985acc953/Projects/Applications/FreeRTOS/FreeRTOS_LoRaWAN_AT/LoRaWAN/App/lora_command.c#L146)
[document](https://www.st.com/resource/en/application_note/dm00346311-examples-of-at-commands-on-icubelrwan-stmicroelectronics.pdf)

Nalan-S, 2025-04-24:
Yes, the UART console shows the same  DevEUI, AppEUI, and JoinEUI as in
device profile. However, after setting
#define LORAWAN_ADR_STATE LORAMAC_HANDLER_ADR_OFF
#define LORAWAN_DEFAULT_DATA_RATE DR_4
I see the following in serial monitor,

###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEui:  2C:F7:F1:20:32:30:8A:0B
###### AppEui:  80:00:00:00:00:00:00:06
###### DevAddr: 26:0D:48:2B
32s831:TX on freq 904100000 Hz at DR 0
54s590:MAC txTimeOut

Here, the data rate is still zero.

Nalan-S, 2025-04-24:
Is this "MAC txTimeout", an error due to hardware issues? Because according
to Seed-Wiki
<https://wiki.seeedstudio.com/LoRa_E5_mini/#develop-with-stm32cube-mcu-package>
the
console should show "MAC txDone".
Can anyone say why "MAC txTimeout" is printed in the serial monitor,
according to the code in the repo.

marekpow, 2025-04-25:
@Nalan-S,

How does it look from the Chirpstack point of view? I asked you about the device profile and the application that uses this profile.
Could you please show me this configuration?

Nalan-S, 2025-04-25:
I don't use Chripstack. I have attached the the device profile from TTN

marekpow, 2025-04-25:
Is the gateway appropriately configured? It's the last thing that must be configured to connect to ATN or any other network.

mingzhangqun, 2025-04-25:
US915 needs to set the region mask, you can reference this function:
hybrid_default_mask(LORAMAC_REGION_US915, 1, true);

uint16_t hybrid_default_mask[REGION_NVM_CHANNELS_MASK_SIZE] = { 0 };
void lorawan_init_default_mask(LoRaMacRegion_t region, uint8_t channel_group, bool valid)
{
    uint16_t *mask = hybrid_default_mask;
    uint8_t ch_max = 8;

    if ((region != LORAMAC_REGION_AU915) && (region != LORAMAC_REGION_US915) &&
        (region != LORAMAC_REGION_CN470)) {
        return;
    }

    if (region == LORAMAC_REGION_CN470) {
        ch_max = 12;
    }

    if (channel_group >= ch_max) {
        mask[0] = 0xFFFF;
        mask[1] = 0xFFFF;
        mask[2] = 0xFFFF;
        mask[3] = 0xFFFF;
        mask[4] = 0x00FF;
        mask[5] = 0x0000;
    }
    else {
        mask[0] = 0x0000;
        mask[1] = 0x0000;
        mask[2] = 0x0000;
        mask[3] = 0x0000;
        mask[4] = 0x0000;
        mask[5] = 0x0000;

        if ((channel_group % 2) == 0) {
            mask[channel_group / 2] = 0x00FF;
        }
        else {
            mask[channel_group / 2] = 0xFF00;
        }

        if (region != LORAMAC_REGION_CN470) {
            mask[4] = (1 << channel_group);
        }
    }
    LOG_D("%s,%d: MASK: 0x%04X, 0x%04X, 0x%04X, 0x%04X, 0x%04X, 0x%04X", __func__, __LINE__,
          mask[0], mask[1], mask[2], mask[3], mask[4], mask[5]);

    if (valid) {
        MibRequestConfirm_t mib_req;
        mib_req.Type = MIB_CHANNELS_DEFAULT_MASK;
        mib_req.Param.ChannelsDefaultMask = hybrid_default_mask;
        LoRaMacMibSetRequestConfirm(&mib_req);

        InitDefaultsParams_t params;
        params.Type = INIT_TYPE_ACTIVATE_DEFAULT_CHANNELS;
        RegionInitDefaults(region, &params);
    }
}

mingzhangqun, 2025-04-27:
@Nalan-S And you can reference the [lorawan-frequency-plans](https://github.com/TheThingsNetwork/lorawan-frequency-plans)

closed 2025-05-09 by mingzhangqun